Unity-Technologies/ROS-TCP-Connector #20: OnApplicationQuit should close the tcp connection

DavidGoedicke, 2020-12-04:
I might got this wrong but after I close the app it does not properly stop the connection and some aspects like moving behavior continues even after the game was stopped. 

Maybe I am doing something wrong but I also cannot see that the connection is closed in the main ``ROSConnection.cs`` script.

Love the package. Thank you all for developing this.

mpinol, 2020-12-07:
Hey David,

Can you please elaborate on the 'moving behavior' you are seeing?
Are you experiencing this in the Editor or have you made a standalone build?
Are you experiencing this with Publishers, Subscribers, or Services?

Thank you for the kind words!

DavidGoedicke, 2020-12-10:
I cannot recreate the issue.

I was publishing the pose from a ZEDmini to ROS and receiving that in Unity to move a virtual object.

I'll close this for now. Should this problems re-apear I'll reopen the issue with more details.

Thanks a gain

closed 2020-12-10 by DavidGoedicke