Unity-Technologies/ROS-TCP-Connector #25: Messages backing up on Unity side

jpapon, 2020-12-14:
Hello,

I've been testing out this plugin, and am having issues with ROS messages getting backed up when they're received on the Unity side. 

I'm publishing out messages from ROS @ 30hz where each message contains vectors of floats at least a few hundred values long. 

What I observe is that the header sequence counter on the Unity side falls behind the ROS side, quickly reaching several seconds behind the ROS side. If I kill the ROS side, frames keep arriving in Unity for several seconds until it clears out what I assume is the TCP buffer.

This is all running on the same machine, so there's no bandwidth limitation. Is this expected behavior? Any idea what might be causing this and where we can start working on the code to fix it?

Thanks!

at669, 2020-12-14:
Hi @jpapon,

I would suggest by starting with checking out the [server](https://github.com/Unity-Technologies/ROS-TCP-Endpoint/blob/main/tcp_endpoint/src/tcp_endpoint/RosTCPServer.py) and [client](https://github.com/Unity-Technologies/ROS-TCP-Endpoint/blob/main/tcp_endpoint/src/tcp_endpoint/RosTCPClientThread.py) scripts in the ROS [tcp_endpoint](https://github.com/Unity-Technologies/ROS-TCP-Endpoint) package. By default, `server_endpoint` will instantiate a TCP server with a buffer size of 1024:

```python
# tcp_endpoint/RosTCPServer.py 
def __init__(self, node_name, buffer_size=1024, connections=10):
```

and the client thread will grab a max of 1024 bytes at a time:

```python
# tcp_endpoint/RosTCPClientThread.py 
while len(data) < full_message_size:
    # Only grabs max of 1024 bytes
    grab = 1024 if full_message_size - len(data) > 1024 else full_message_size - len(data)
    packet = self.conn.recv(grab)

    if not packet:
        print("No packets...")
        break

    data += **packet**
```

Increasing the value from 1024 might reduce this behavior. Let us know if this helps!

jpapon, 2020-12-15:
Hi @at669 , thanks for the quick response!
I've tried various buffer sizes (4k, 8k, 32k etc...) to no avail, the buffer keeps filling up.
It looks like the Unity side just isn't clearing messages fast enough - if I lower my publisher's rate to 10hz, the backup goes away. Also, when I hit play in the editor to stop my sim, the console shows it clearing out all my backed up messages, and they're getting decoded too slowly to match the required rate.

Using a prerecorded bagfile and adjusting playback rate using the `-r` option, it appears that the problem occurs whenever my ROS message publish rate is higher than my Unity framerate.

Is the Unity side reading a maximum of one message (per topic) per frame, and letting additional messages accumulate in the TCP buffer?

jpapon, 2020-12-15:
Okay, I've verified this is indeed the problem by setting `Application.targetFrameRate`. Whatever value I set that to is the maximum frame rate my messages can be received before they start backing up.

Going to dig into the RosConnection.cs code a bit to see if I can figure out where it's only reading one message per topic per Unity frame.

jpapon, 2020-12-15:
@at669 I have also just confirmed that in fact RosConnection.cs only handles one incoming message per frame, period. So the behavior for multiple subscribers is probably not what people expect, and will only work well if Unity is running at high frame rates.

My understanding of C#/Unity await & Task.Yield() is still poor though, so I can't figure out why it is only calling ReadMessage once per frame.

LaurieCheers-unity, 2020-12-15:
Oh, interesting. Apparently that's just how await is implemented in Unity.

My theory is that after handling the first message, we can do a second loop, `while(tcpListener.Pending()) {...}` and just synchronously clean out the message queue before going back to the await. Currently testing this.

jpapon, 2020-12-15:
@LaurieCheers-unity Yup this seems to work clearing off all messages, nice catch!
```
while (tcpListener.Pending())
{
   tcpListener.AcceptTcpClient()
}
```

LaurieCheers-unity, 2020-12-15:
Heh, well if you do that you're just dropping messages on the floor. Try:

                    while(tcpListener.Pending())
                    {
                        tcpClient = tcpListener.AcceptTcpClient();
                        task = StartHandleConnectionAsync(tcpClient);
                        if (task.IsFaulted)
                            await task;
                    }

jpapon, 2020-12-15:
Yeah, I edited it as you were typing that =) Yours works for me too.

LaurieCheers-unity, 2020-12-16:
This fix is now in dev branch, will be merged into main soon.

closed 2020-12-16 by LaurieCheers-unity