Unity-Technologies/ROS-TCP-Connector #33: Closing the TCP connection when not playing

garycoding, 2020-12-18:
Hi, I've been using this for a couple of weeks now and it's been great. Thank you for creating it!

I've recently run into this problematic behavior with the connection and made a fix for it.

Currently, when the Application quits (when the user stops playing the scene in the editor), the TcpListener in RosConnection.cs continues to process incoming messages and call registered subscribers.

This is a problem if one has stopped playing in the editor and then does a "Build and Run". In this case the running application will throw an exception and not connect to the ROS server because the TcpListener which was started in the editor has not released the address.

Here's a fix that works for me. It's checking Application.isPlaying in the outer loop of StartMessageServer() in RosConnection.cs to prevent the code from creating a TcpListener when the application is not playing.
And, it's checking Application.isPlaying in the inner loop so that it can call Stop() the TcpListener, set it to null and break out of the inner loop.

Also, I think this is related to https://github.com/Unity-Technologies/ROS-TCP-Connector/issues/20

Here's the modified version of StartMessageServer():

`    protected async void StartMessageServer(string ip, int port)
    {
        if (alreadyStartedServer)
            return;

        alreadyStartedServer = true;
        while (true)
        {
            if (!Application.isPlaying)
            {
                await Task.Yield();
            }
            else
            {
                TcpListener tcpListener;
                try
                {
                    tcpListener = new TcpListener(IPAddress.Parse(ip), port);
                    tcpListener.Start();
                }
                catch (Exception e)
                {
                    Debug.LogError("Exception raised!! " + e);
                    return;
                }

                Debug.Log("ROS-Unity server listening on " + ip + ":" + port);

                try
                {
                    while (true)   //we wait for a connection
                    {
                        if (!Application.isPlaying)
                        {
                            if (tcpListener != null)
                            {
                                tcpListener.Stop();
                                tcpListener = null;
                                await Task.Yield();
                            }
                            break;
                        }

                        var tcpClient = await tcpListener.AcceptTcpClientAsync();

                        var task = StartHandleConnectionAsync(tcpClient);
                        // if already faulted, re-throw any error on the calling context
                        if (task.IsFaulted)
                            await task;

                        while (tcpListener.Pending())
                        {
                            tcpClient = tcpListener.AcceptTcpClient();
                            task = StartHandleConnectionAsync(tcpClient);
                            if (task.IsFaulted)
                                await task;
                        }
                    }
                }
                catch (Exception e)
                {
                    Debug.LogError("Exception raised!! " + e);
                }
            }
        }
	}
`

garycoding, 2020-12-18:
My code paste above was cut off. Here's the whole function:

`
    protected async void StartMessageServer(string ip, int port)
    {
        if (alreadyStartedServer)
            return;

        alreadyStartedServer = true;
        while (true)
        {
            if (!Application.isPlaying)
            {
                await Task.Yield();
            }
            else
            {
                TcpListener tcpListener;
                try
                {
                    tcpListener = new TcpListener(IPAddress.Parse(ip), port);
                    tcpListener.Start();
                }
                catch (Exception e)
                {
                    Debug.LogError("Exception raised!! " + e);
                    return;
                }

                Debug.Log("ROS-Unity server listening on " + ip + ":" + port);

                try
                {
                    while (true)   //we wait for a connection
                    {
                        if (!Application.isPlaying)
                        {
                            if (tcpListener != null)
                            {
                                tcpListener.Stop();
                                tcpListener = null;
                                await Task.Yield();
                            }
                            break;
                        }

                        var tcpClient = await tcpListener.AcceptTcpClientAsync();

                        var task = StartHandleConnectionAsync(tcpClient);
                        // if already faulted, re-throw any error on the calling context
                        if (task.IsFaulted)
                            await task;

                        while (tcpListener.Pending())
                        {
                            tcpClient = tcpListener.AcceptTcpClient();
                            task = StartHandleConnectionAsync(tcpClient);
                            if (task.IsFaulted)
                                await task;
                        }
                    }
                }
                catch (Exception e)
                {
                    Debug.LogError("Exception raised!! " + e);
                }
            }
        }
	}
`

at669, 2020-12-18:
Hi @garycoding, thanks for the feedback! We'll take a look into this issue further and add in your suggestions in a future update.

LaurieCheers-unity, 2020-12-22:
Not quite the same fix, but this should now be fixed in main branch.

closed 2020-12-22 by LaurieCheers-unity