Unity-Technologies/ROS-TCP-Connector #35: Unity as the service node

orkop465, 2020-12-22:
Hi, for the current project I'm working on I need Unity to act as the service node, able to receive requests and send responses to ROS. I want to be able to call services using the service caller in RQT. This is something I was able to do with ROSSharp, however that functionality doesn't seem to exist in ROS-TCP-Connector. According to my understanding, the Unity service would need to advertise itself to ROS in order to be stored in the list of services, which can be called from the service caller. The SendServiceMessage function in the ROSConnection script seems similar to what I need however the other way around. Is there any way I can make a ReceiveServiceMessage function using the given tools? Thanks.

vidurvij-Unity, 2020-12-22:
Hi @orkop465,
Thanks for bringing up this issue. We are aware of this issue and will be working on adding this feature next year.
In the meantime, you can achieve the same behavior by publishing messages to the topic Unity subscribes too and publishing a response message from Unity in a different topic.

I hope this answers your question. Feel free to reach out if you have any further questions
Vidur

LaurieCheers-unity, 2020-12-23:
Hi, I've checked in an early work-in-progress implementation of this in https://github.com/Unity-Technologies/ROS-TCP-Connector/tree/laurie/UnityService and https://github.com/Unity-Technologies/ROS-TCP-Endpoint/tree/laurie/UnityService.

On the ROS side, you declare a UnityService in your server_endpoint dict:

        'topic': UnityService('topic', SrvType, tcp_server)

And on the Unity side, it works like a subscriber - your class should call:

        ros.ImplementService<ServiceRequestType>("topic", ServiceCallback);

The ServiceCallback function should implement your service behaviour, taking a RequestType argument and returning a Message.
This code worked in early tests but is still in development, let me know if you hit any problems.

orkop465, 2020-12-30:
Hey, sorry for the late response. I added your branch as the package to my project, however now I'm getting some other errors, namely these two:

ROS-TCP-Connector/Runtime/ROSGeometry/ROSVector3.cs(202,104): error CS1501: No overload for method 'ToString' takes 2 arguments
ROS-TCP-Connector/Runtime/ROSGeometry/ROSQuaternion.cs(145,95): error CS1501: No overload for method 'ToString' takes 2 arguments

This didn't happen with the main branch, which is strange. Also when I went into the scripts and changed the code from
internalVector.ToString(format, formatProvider);
to
internalVector.ToString(format);

I started receiving more errors related to types or namespaces not existing, specifically 'TcpEndpoint', 'RosUnityHandshakeResponse', and 'RosUnityError', all coming from the ROSConnection script.

Any idea as to how to fix these bugs? Maybe something went wrong when adding the package? Thanks!

LaurieCheers-unity, 2021-01-04:
Hmm, not obvious what went wrong here.

What version of Unity are you using? (2020.1.14f1 works for me.)

What did you write in the package manager to add the package? (
https://github.com/Unity-Technologies/ROS-TCP-Connector.git#laurie/UnityService
works for me.)

orkop465, 2021-01-11:
Hey, seems to have been a version mismatch, my apologies. I'll check now about the services and get back to you very soon. Thanks for your help so far!

orkop465, 2021-01-11:
Hello again, just tried it out with a service and it works great! Thanks a ton for your help, I'll close this issue now.

closed 2021-01-11 by orkop465