Unity-Technologies/ROS-TCP-Connector #39: Logic error when receiving data Unity Side

P3TE, 2021-01-06:
Hey there, I noticed an issue with the logic of receiving data on the unity side in the ROSConnection

dev branch
Line 216
and the issue also occurs on line 125 (duplicated code).

`

byte[] readBuffer = new byte[full_message_size];
int numberOfBytesRead = 0;

while (networkStream.DataAvailable && numberOfBytesRead < full_message_size)
{
    int bytesRead = networkStream.Read(readBuffer, 0, readBuffer.Length);
    offset += bytesRead;
    numberOfBytesRead += bytesRead;
}

`

The problem is that if the entire message isn't completely read in one Read, the next read will override the bytes read. Also it has the potential to read data from the next message.
Anyways, a fix could be something like this:

`

byte[] readBuffer = new byte[full_message_size];
int numberOfBytesRead = 0;
int bytesRemaining = full_message_size;
while (networkStream.DataAvailable && bytesRemaining > 0)
{
    int bytesRead = networkStream.Read(readBuffer, numberOfBytesRead, bytesRemaining);
    offset += bytesRead;
    numberOfBytesRead += bytesRead;
    bytesRemaining -= bytesRead;
}

`

Also I found that the 
`
callback(msg);
`
Should probably have it's own try catch as if one subscriber throws an exception due to poor code subsequent subscribers won't receive the message. I mean ideally the callback(msg); should never throw an exception, but having it's own try catch would reduce the impact when a bug does occur and there are multiple subscribers for the same topic.

Oh and while I'm here, the same read code is used twice for both subscribers and services so it's really the first issue propagated twice so it could be improved if it uses the same code.

P3TE, 2021-01-06:
I actually fixed this issue on my fork, let me know if you would like me to create a pull request from a cherry pick of these two commits or something.
You can see the two commits here:
https://github.com/P3TE/ROS-TCP-Connector/commit/fe4801b7b2de32005d58ff52230970a170a46e38
Also:
https://github.com/P3TE/ROS-TCP-Connector/commit/24ab103e5afe3cc6439367c09897be4073ce580d

LaurieCheers-unity, 2021-01-11:
Good catch, thanks! I'll fix this.

closed 2021-01-20 by LaurieCheers-unity