Unity-Technologies/ROS-TCP-Connector #58: package.json missing

IdeaMogul, 2021-02-01:
[This](https://github.com/Unity-Technologies/ROS-TCP-Connector/pull/57) commit deleted package.json. Can no longer import git repo into Unity.

at669, 2021-02-01:
Hi @IdeaMogul,

package.json is now located under [com.unity.robotics.ros-tcp-connector/package.json](https://github.com/Unity-Technologies/ROS-TCP-Connector/blob/dev/com.unity.robotics.ros-tcp-connector/package.json) on the `dev` branch. We recommended using the latest release (v0.1.2) for the most stable version, using `https://github.com/Unity-Technologies/ROS-TCP-Connector.git#v0.1.2`.

If you would like to proceed with using the `dev` branch, you can import the project using its specific subfolder.  Using the same `Add package from git URL` option, you can enter the following: `https://github.com/Unity-Technologies/ROS-TCP-Connector.git?path=/com.unity.robotics.ros-tcp-connector`.

Let me know if these options resolve your issue. Thanks!

IdeaMogul, 2021-02-02:
Got it. Thanks, Amanda!

closed 2021-02-02 by IdeaMogul