Unity-Technologies/ROS-TCP-Connector #71: Installation of ROS-TCP-Connector fails via Unity PackageManager due to missing package manifest file

safdark, 2021-03-12:
Environment:
- Unity 2021.1b
- MacOS Catalina (10.15.7)

As per the installation instructions, I attempted installing the following git URL (via the Unity Package Manager):
https://github.com/Unity-Technologies/ROS-TCP-Connector.git

However, the following errors show up:

```
[Package Manager Window] Cannot perform upm operation: Unable to add package [https://github.com/Unity-Technologies/ROS-TCP-Connector.git]:
  [https://github.com/Unity-Technologies/ROS-TCP-Connector.git] does not point to a valid package repository. No package manifest was found. Verify the repository URL and make sure the package is located in the root folder of the repository. [NotFound].
UnityEditor.EditorApplication:Internal_CallUpdateFunctions () (at /Users/bokken/buildslave/unity/build/Editor/Mono/EditorApplication.cs:327)

[Package Manager Window] Error adding package: https://github.com/Unity-Technologies/ROS-TCP-Connector.git.
UnityEditor.EditorApplication:Internal_CallUpdateFunctions () (at /Users/bokken/buildslave/unity/build/Editor/Mono/EditorApplication.cs:327)
```

safdark, 2021-03-12:
Had logged this under Unity-Robotics-Hub earlier, but moved to ROS-TCP-Connector since it seemed the more appropriate target.

safdark, 2021-03-12:
Works when using (https://github.com/Unity-Technologies/ROS-TCP-Connector.git#v0.1.2) as git URL. So, closing.

mrpropellers, 2021-03-12:
Hi @safdark thanks for bringing this to our attention!  It looks like the instructions in that guide are stale and need to be updated. Right now the most recent tagged release of ROS-TCP-Connector is v0.2.0, but we've changed the directory structure in the repository to accommodate automated testing, and the package json is no longer at the root of the repository (hence the error).  To use v0.2.0, you'll want to use a url like this:  
`https://github.com/Unity-Technologies/ROS-TCP-Connector.git?path=/com.unity.robotics.ros-tcp-connector#v0.2.0`  
I'll go ahead and submit updates to the quick_setup and see if there aren't other places that need similar changes.

mrpropellers, 2021-03-12:
Inspecting a little closer, this issue is already addressed on `dev`, but we haven't pushed the updated instructions to `main` yet.  I've added similar instructions to this repo's readme as well, but I'll keep this issue open for visibility until we've merged the new instructions into main on the hub.

peifeng-unity, 2021-04-13:
Closing this issue since it is inactive for a while. @safdark Please feel free to reopen it if necessary.

closed 2021-04-13 by peifeng-unity

maycolv, 2023-01-17:
Please, help me because I can´t install 
Version:  Unity 2021.3.16f1 (free)
SO: Windows 10, 64 bits
Package intalled sucessfull: com.unity.scripting.pythom

I trying install it but alway have error, It doesn't work with #v0.6.0 #v0.2.0.
https://github.com/Unity-Technologies/ROS-TCP-Connector.git?path=/com.unity.robotics.ros-tcp-connector
https://github.com/Unity-Technologies/ROS-TCP-Connector.git?path=/com.unity.robotics.ros-tcp-connector#v0.7.0

error
[Package Manager Window] Error adding package: https://github.com/Unity-Technologies/ROS-TCP-Connector.git?path=/com.unity.robotics.ros-tcp-connector#v0.7.0