Unity-Technologies/ROS-TCP-Connector #85: tf broadcast

silasalves, 2021-04-20:
I am writing a mobile robot base software and could successfully move it using ROS Publishers and Subscribers. Now I am implementing odometry (following [this tutorial](http://wiki.ros.org/navigation/Tutorials/RobotSetup/Odom)) and I realized that ROS-TCP-Connector does not seem to support [tf](http://wiki.ros.org/tf) or [tf2](http://wiki.ros.org/tf2).

Is there any plan to add support to tf on ROS-TCP-Connector anytime soon? Or can you give me a suggestion on a workaround?

Thanks a lot!

LaurieCheers-unity, 2021-04-20:
Hi, thanks for asking! This feature is in development and we hope to release it sometime this quarter.

silasalves, 2021-04-20:
Thanks, that is great news!

For now, I will create a common subscriber on ROS side and broadcast that information on tf.

closed 2021-04-28 by silasalves