Unity-Technologies/ROS-TCP-Connector #92: Revert the M-Prefix class renaming

JimWest, 2021-04-22:
**Is your feature request related to a problem? Please describe.**
I was using the ROS-TCP-Connector in a project and wrote quite some publisher and subscribers, now I wanted to upgrade to the newest version and the naming of all Messages were changed and I would have to change several files. It also makes the usage inconvenient, if I have an compressed Image and want to change its header, the variable is still named header, but the class is now `MHeader`. A Prefix to a class also violates all the C# / .Net naming guidelines (which are also mentioned in the Contribution Guidelines).

**Describe the solution you'd like**
Revert the change and remove the M-Class-Prefix.

**Describe alternatives you've considered**
If you want to show that those Message-Classes are autogenerated or something like that, move them to a new namespace (like RosMessageTypes.Sensor.Generated).

LaurieCheers-unity, 2021-04-23:
Hi, as I explained in [the PR](https://github.com/Unity-Technologies/ROS-TCP-Connector/pull/66#issuecomment-824701120), the M prefix is there for a reason. I discussed it with the team and we agreed a "Msg" suffix (e.g. class Vector3Msg) might have done a good job of solving the problem while also satisfying coding guidelines; so we may consider changing to that style at some point, but changing again so soon doesn't seem worth the extra inconvenience, so we're keeping the prefix for now.

I'm sorry that you have to change your code, but I remind you this is an experimental package - some instability is to be expected.

closed 2021-04-23 by LaurieCheers-unity

JimWest, 2021-04-28:
Mhm alright, sorry for my rage Issue ;-)