Unity-Technologies/ROS-TCP-Connector #93: ROS Time

silasalves, 2021-04-22:
**Is your feature request related to a problem? Please describe.**
I just realized that ROS-TCP-Connector does not fill in the `stamp` field of the [Header](http://docs.ros.org/en/api/std_msgs/html/msg/Header.html), so all stamped messages are submitted without time information.

**Describe the solution you'd like**
Either synchronize Unity with ROS (maybe using NTP) or generating the [simulation time](http://wiki.ros.org/Clock) and publishing it to `/clock`.

**Describe alternatives you've considered**
Right now I am implementing both cases. I installed an NTP server on Ubuntu 18.04 (ROS Melodic) and adapted this [simple NTP client](https://stackoverflow.com/a/20157068) to synchronize Unity and ROS.

For scenarios where the simulation time can be used, I publish [rosgraph_msgs/Clock](http://docs.ros.org/en/api/rosgraph_msgs/html/msg/Clock.html) messages to `/clock`.

**Additional context**
ROS [time](http://docs.ros.org/en/api/rostime/html/classros_1_1Time.html) is represented by two components: `secs` for seconds and `nsecs` for nanoseconds. I have been using Unity's `Time.timeAsDouble` to generate the ROS time, but I wonder if that is the best way to do it.

mrpropellers, 2021-04-26:
Hey @silantzis thanks for the detailed request! Time Management is absolutely in our sights, but it is quite tricky to implement a system that respects both all the expectations/assumptions made by ROS applications, which can use either `/clock` time or OS wall time, and Unity systems, which often assume OS wall-clock is the definitive time source.  Note that even with such a feature in place, the tcp connector will still not auto-populate headers (as this isn't the expected behavior for a ROS application), so that's still something you'll need to write context-specific logic for.  If you are fine with using Unity's host OS's wall clock as the time source, `Time.timeAsDouble` is a fine interface for fetching this time. We won't have a robust system in place for managing sim time for a while, though, if you need to use the time from `/clock` today, unfortunately you'll have to manually subscribe to `/clock` and write an implementation that suits your use case.

silasalves, 2021-04-28:
Thanks, @mrpropellers, for the explanation! You are right, the ROS API does not auto-populate the `stamp` field and I agree that Unity's implementation should do the same.

For now, I will use my current implementation that so far seems to be working fine. I am using `Time.fixedTimeAsDouble` when using unity as the source of the `/clock` messages, and `Time.realtimeSinceStartup` when using OS's wall clock synchronized with NTP.

closed 2021-04-28 by silasalves