Unity-Technologies/ROS-TCP-Connector #97: Website Link Issue

A-Ivan, 2021-05-05:
On the Main branch of the ROS-TCP-Connector (https://github.com/Unity-Technologies/ROS-TCP-Connector), in the Tutorials section.

**Tutorials**
ROSConnection: See the **Unity Robotics Hub** repository for information and tutorials on how to use this component.

This **Unity Robotics Hub** has an incorrect link, it links to "https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/**master**/tutorials/ros_unity_integration/README.md" but I believe it should link to "https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/**main**/tutorials/ros_unity_integration/README.md", as there is no **master** branch.

at669, 2021-05-05:
Hi @A-Ivan, I've linked a small PR for this fix above, feel free to use that link to keep updated with this change. Thanks for catching this!

A-Ivan, 2021-05-05:
Great. You're welcome.

at669, 2021-05-05:
As the fix has been merged to `dev`, I'll now close out this issue. Thanks again!

closed 2021-05-05 by at669