Unity-Technologies/ROS-TCP-Connector #121: ros_tcp_endpoint: Cannot locate rosdep definition for [message_runtime]

neuralmimicry, 2021-06-11:
**Describe the bug**
ros_tcp_endpoint: Cannot locate rosdep definition for [message_runtime]

**To Reproduce**
Clean install of ROS2 Foxy
git clone the Unity-Technologies/ROS-TCP-Connector
Move ROS-TCP-Connector folder to foxy/src folder
rosdep install --from-paths src --ignore-src --rosdistro foxy -y --skip-keys "console_bridge fastcdr fastrtps rti-connext-dds-5.3.1 urdfdom_headers"

**Console logs / stack traces**
```
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
ros_tcp_endpoint: Cannot locate rosdep definition for [message_runtime]
```

**Expected behavior**
Expected to have the ROS-TCP-Connector installed

**Environment (please complete the following information, where applicable):**
- Unity Version: Unity 2020.3.11f1
- Unity machine OS + version: RHEL8.4
- ROS machine OS + version: Ubuntu 18.04, ROS Foxy
- ROS–Unity communication: kvm

LaurieCheers-unity, 2021-06-14:
ROS-TCP-Endpoint currently supports only ROS 1 (melodic or noetic).
ROS2 (foxy) support should be coming soon.

sdiao, 2021-08-02:
Hi @neuralmimicry, We are proud to announce that our new release now officially supports ROS2. This version has a lot of bug fixes and improvements over the alpha, so we recommend switching to this version.

You can get started by following the tutorial here:
https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/ros_unity_integration/publisher.md

I'm going to close this issue now. Please feel free to reach out if you have any question using ROS2. Thanks.

closed 2021-08-02 by sdiao