Unity-Technologies/ROS-TCP-Connector #132: Standard on whether .meta files should be added to the git repo.

P3TE, 2021-07-14:
Hi, not really a feature request, but after adding this repo as a submodule to a git project, currently on the dev branch, I have found that Unity (Using 2021.1.14f1) has automatically generated a number of .meta files that aren't covered by the .gitignore.

I just would like to know what the standard here is, should all .meta files be .gitignored, if not which ones should be added?

For further reference, here are all the .meta files Unity has added but are not tracked.
<details>
<summary>[Expandable] .meta files added but not tracked:</summary>
<p>

- CODE_OF_CONDUCT.md.meta
- CONTRIBUTING.md.meta
- InfoAcknowledgements.md.meta
- LICENSE.meta
- MessageGeneration.md.meta
- README.md.meta
- ROSGeometry.md.meta
- TestRosTcpConnector.meta
- TestRosTcpConnector/Assets.meta
- TestRosTcpConnector/Packages.meta
- TestRosTcpConnector/Packages/manifest.json.meta
- TestRosTcpConnector/Packages/packages-lock.json.meta
- TestRosTcpConnector/ProjectSettings.meta
- TestRosTcpConnector/ProjectSettings/AudioManager.asset.meta
- TestRosTcpConnector/ProjectSettings/ClusterInputManager.asset.meta
- TestRosTcpConnector/ProjectSettings/DynamicsManager.asset.meta
- TestRosTcpConnector/ProjectSettings/EditorBuildSettings.asset.meta
- TestRosTcpConnector/ProjectSettings/EditorSettings.asset.meta
- TestRosTcpConnector/ProjectSettings/GraphicsSettings.asset.meta
- TestRosTcpConnector/ProjectSettings/InputManager.asset.meta
- TestRosTcpConnector/ProjectSettings/NavMeshAreas.asset.meta
- TestRosTcpConnector/ProjectSettings/PackageManagerSettings.asset.meta
- TestRosTcpConnector/ProjectSettings/Physics2DSettings.asset.meta
- TestRosTcpConnector/ProjectSettings/PresetManager.asset.meta
- TestRosTcpConnector/ProjectSettings/ProjectSettings.asset.meta
- TestRosTcpConnector/ProjectSettings/ProjectVersion.txt.meta
- TestRosTcpConnector/ProjectSettings/QualitySettings.asset.meta
- TestRosTcpConnector/ProjectSettings/TagManager.asset.meta
- TestRosTcpConnector/ProjectSettings/TimeManager.asset.meta
- TestRosTcpConnector/ProjectSettings/URPProjectSettings.asset.meta
- TestRosTcpConnector/ProjectSettings/UnityConnectSettings.asset.meta
- TestRosTcpConnector/ProjectSettings/VFXManager.asset.meta
- TestRosTcpConnector/ProjectSettings/VersionControlSettings.asset.meta
- TestRosTcpConnector/ProjectSettings/XRSettings.asset.meta
- Third Party Notices.md.meta
- com.unity.robotics.ros-tcp-connector.meta
</p>
</details>

peifeng-unity, 2021-07-14:
Hello @P3TE, thanks for your feedback. I understand your issue and the reason that cause it. In our repository, the Unity package is in [the `com.unity.robotics.ros-tcp-connector` subfolder of the repo](https://github.com/Unity-Technologies/ROS-TCP-Connector/tree/main/com.unity.robotics.ros-tcp-connector) and it is the only folder that needs to be put into the Unity project. So you will find all files under the `com.unity.robotics.ros-tcp-connector` folder contain `.meta` files, but there are no `.meta` files outside this folder. 

If you want to import the ROS-TCP-Connector package, the easiest way would be to install it via the package manager as described in [the README.md](https://github.com/Unity-Technologies/ROS-TCP-Connector#installation).

Would you mind letting us know your use case? We are interested to learn the reason that you wish to use this repo as a submodule in your git repository?

P3TE, 2021-07-15:
Hi @peifeng-unity, my current use case is that I'm creating a simulator in Unity for ROS and I want to be be on the bleeding edge of development so I can get the latest updates and possibly even contribute to the repo with changes.

The simulator I'm working on aims to be compatible with both ROS1 and ROS2.

The reason it's currently a git submodule is that I have setup a git repo for the simulator I'm working on and I wanted to add this repo to the Assets directory within the simulator git project, so added this project as a git submodule.

peifeng-unity, 2021-07-15:
Thank you, @P3TE! It is really good to know your use case. Yeah, we usually use the same workflow as what you described when developing the package. I have created an internal ticket AIRO-992 to follow up and we probably can make the instructions clearer

peifeng-unity, 2021-07-19:
I am going to close this issue since it has been resolved. We will follow up and update the workflow in our tutorials.

closed 2021-07-19 by peifeng-unity