Unity-Technologies/ROS-TCP-Connector #133: CameraInfo Generator

P3TE, 2021-07-14:
**Is your feature request related to a problem? Please describe.**
In ROS, when working with [Image](http://docs.ros.org/en/melodic/api/sensor_msgs/html/msg/Image.html) messages (particularly stereo cameras), the [CameraInfo](http://docs.ros.org/en/noetic/api/sensor_msgs/html/msg/CameraInfo.html) topic is important for nodes such as [stereo_image_proc](http://wiki.ros.org/stereo_image_proc) and [visual odometery](http://wiki.ros.org/viso2_ros). For the past two projects I've written some code to automatically generate CameraInfo for unity cameras. I was considering writing a more general version and adding the code to the ROS-TCP-Connector.

**Describe the solution you'd like**
Add a CameraInfo generator based on Unity Camera inputs.

**Describe alternatives you've considered**
Re-writing the code for every project.

**Additional context**
I understand that this project is mainly focused on the Connection between ROS and Unity, so maybe this is more of a utility that should go into a utility repository that's Unity ROS related, but not specifically the connector, does Unity have one for such things?
Note: I checked [Unity-Robotics-Hub](https://github.com/Unity-Technologies/Unity-Robotics-Hub), but didn't see anything suitable.

peifeng-unity, 2021-07-14:
Hello @P3TE, thanks for your feedback. I created an internal ticket AIRO-989 and we will follow up on it.

peifeng-unity, 2021-07-19:
Hi @P3TE, our team had a discussion and we think this is a good feature to have. This repository might be the best place for the generator. It may take us some time before we come back and do the work. So, if you'd like to implement this feature, we are more than happy to help to merge your changes to the codebase.

P3TE, 2021-07-19:
I'll have a crack at implementing this and create a pull request when I'm done.

peifeng-unity, 2021-07-20:
Awesome! Thanks

P3TE, 2021-07-20:
Hey, I've created a [pull request](https://github.com/Unity-Technologies/ROS-TCP-Connector/pull/142).

There are a couple of features that could be added that might be nice to haves, but I didn't need them for my work.

They include the nice to haves:
- Camera Resolution Scaling.
- Non Zero Vertical Camera Offset. (Currently only horizontal camera offsets are supported, for stereo cameras)
- Lens Shift. (A feature of the Physical Camera in Unity)

Also kind of nice to have, but harder to implement / less useful:
 - Distortion Models. (Would require also making some post processing effect shaders to produce the distortion in the first place)
 - Rectification Matrix (Other than the identity).
 - Axis Skew.

sdiao, 2021-08-02:
The PR was merged. Close this issue now.

closed 2021-08-02 by sdiao