Unity-Technologies/ROS-TCP-Connector #141: Retrieve topic message types

bryceikeda, 2021-07-18:
I see in the RosConnection script there is a function that retrieves the topic names from the server_endpoint. I was wondering if it would be possible to add a feature/function that also allows me to retrieve topic message types. Similarly, adding a way to get if they are subscribers or publishers.

peifeng-unity, 2021-07-19:
Thanks for the request, @bryceikeda, I created an internal ticket AIRO-1005 for this feature. We will follow up and let you know.

bryceikeda, 2021-07-19:
@peifeng-unity Thank you!

LaurieCheers-unity, 2021-07-20:
Hi, in the new version 0.5.0 you can call the function ROSConnection.instance.GetTopicAndTypeList().
https://github.com/Unity-Technologies/ROS-TCP-Connector/blob/f7f669f212423ff03817a2094f489547158a3852/com.unity.robotics.ros-tcp-connector/Runtime/TcpConnector/ROSConnection.cs#L193

bryceikeda, 2021-07-21:
@LaurieCheers-unity Oh awesome. I had problems with v0.5.0 so I didn't see this function. Thank you for getting back to me.

closed 2021-07-21 by bryceikeda