Unity-Technologies/ROS-TCP-Connector #171: Compile Errors when Switching to WebGL

frankjoshua, 2021-09-02:
Some of the method and class signatures change when switching unity platform to WebGL.
For example:
public TimeMsg(int sec, uint nanosec);
becomes
public TimeMsg(uint sec, uint nanosec);

peifeng-unity, 2021-09-07:
Hi @frankjoshua, thank you for bringing it up. I have created an internal ticket AIRO-1214 and we will follow up on it.

mrpropellers, 2021-09-14:
This looks like the difference between ROS and ROS2's Time messages. @frankjoshua, can you confirm you did not switch to ROS from ROS2 in the ROS Settings panel when switching to WebGL? If you didn't manually do this, maybe some process when changing the platform target also reset out preprocessor define.

frankjoshua, 2021-09-15:
I checked. It's set to ROS2 in the Robotics-> ROS Settings menu. 
I tried to toggle it back to ROS1 the to ROS2. But it wouldn't switch due to the compile error.
Is there a config file or something to check?

mrpropellers, 2021-09-15:
Under Edit -> Project Settings -> Player -> Script Compilation you should see a `ROS2` define, which forces the connector package to compile the ROS2 version of the message definitions. If there is no `ROS2` there, it will compile for ROS(1) by default.  If the `ROS2` is already there, then there is something strange going on...

frankjoshua, 2021-09-16:
ROS2 was missing from the Scripting Define symbols. I added it and it fixed the compile errors. However my ROS messages are not compiling.
[image]

mrpropellers, 2021-09-27:
Hey @frankjoshua sorry I disappeared on vacation for a bit but looking at this, my next guess is that there are some older message definitions inside the local project's RosMessages directory. I think at some point we added TFMessage to the collection of messages we build by default inside ROS-TCP-connector, so they may be getting built superfluously inside your version of the Nav2-SLAM-Example.  Can you try deleting the TF2 subdirectory out of your project assets folder? If that doesn't fix it outright, you can re-import the project's local copies of the tf2 .msg files to generate new versions of the definitions that should be compatible with the version of ROS-TCP-Connector you're on.

frankjoshua, 2021-09-28:
I deleted the messages and reimported. That got me a little further. Now there is an error about GUItility.cs

This is possible to do right? I have not modified any code. I'm just running the Nav2 Slam example and trying to run the simulation in a browser.
[image]

mrpropellers, 2021-09-28:
I can't say for certain whether building our packages for WebGL works, as its not one of our supported platforms, but I'm not aware of anything that would keep it from working, either. 

The Robotics-Nav2-SLAM-Example itself is not built into a Player as part of our CI/CD pipeline yet, though, so you may be hitting two flavors of complications at the same time:
1. Getting Robotics-Nav2-SLAM-Example buildable into a Player (may require a little re-organization to compartmentalize the Editor-only code)
2. Getting the Robotics packages, in general, built for WebGL (may or may not be possible)

closed 2021-11-20 by github-actions[bot]