Unity-Technologies/ROS-TCP-Connector #183: Parameter service

Raphtor, 2021-09-17:
**Is your feature request related to a problem? Please describe.**
I want to be able to set/get parameters for a Unity ROS node. I think it can be done from scratch using the service API in ROS2, but it would be nice to have an easy interface.

**Describe the solution you'd like**
A generalized class that allows this to be done easily and arbitrarily for all nodes.

**Describe alternatives you've considered**
I think its possible to use ROS2 services to manually create this ([ROS2 parameters are implemented as services](https://docs.ros.org/en/foxy/Tutorials/Services/Understanding-ROS2-Services.html#ros2-service-list))

**Additional context**
N/A

sdiao, 2021-09-23:
@Raphtor , thanks for reaching out. We are currently reviewing your feature request.

sdiao, 2021-10-07:
Tacked internally in AIRO-1281

Raphtor, 2021-10-07:
> @Raphtor , could you share a bit more about your use case? Are you currently using ROS1 or ROS2?

I am currently using ROS2. My use case is that I have a bunch of nodes that should share parameters. Particularly, I’m using Unity as a simulator/display for multiple simulated/real robots. Each robot has its own parameters and nodes, and the nodes running in Unity should share that information. I am currently solving this by just reading a shared parameter file, but this is inefficient since now all the robots and the computer running unity have to have the correct file.

hyounesy, 2022-01-25:
[AIRO-1281]

junaidali123, 2024-02-13:
Hi, is there any update on this?