Unity-Technologies/ROS-TCP-Connector #202: Unable to install ros-tcp-visualizer

fangkd8, 2021-10-03:
I just follow the tutorial from https://github.com/Unity-Technologies/ROS-TCP-Connector/blob/v0.6.0/com.unity.robotics.visualizations/Documentation~/README.md, but my unity continues reporting errors, just shown in the image. 

[image]

LaurieCheers-unity, 2021-10-04:
Hi, you'll need to also install the ROS TCP Connector.

I'll update that doc.

fangkd8, 2021-10-04:
But before installing visualizer package, I already installed tcp connect with the exact link you shared through package manager

LaurieCheers-unity, 2021-10-04:
Interesting. What version of Unity are you using? Does your package manager look like this?
[image]

fangkd8, 2021-10-05:
> Interesting. What version of Unity are you using? Does your package manager look like this? [image]

Exactly like this, my unity version is 2020.3.19f1 on ubuntu 18.04.

LaurieCheers-unity, 2021-10-05:
Strange... I don't have an explanation for these symptoms. Could you provide a copy of your editor log, showing your compile errors? (You can find it by going to the three dots menu at the top right corner of your Console window, and selecting "Open Editor Log".)

closed 2021-11-03 by github-actions[bot]