Unity-Technologies/ROS-TCP-Connector #232: Display data from stereo camera

Russ76, 2021-11-07:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
The Intel D435 sends different image frames for left and right camera. How do we display these in the Unity scene for the Oculus Quest? Would this be termed 2d or 3d?

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
It won't be hard to display both images side by side, but is this what we want to do in order to send the video to each eye in the headset? The human viewer makes it into 3d!

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
I want it to seem like the headset wearer is "inside" the robot, seeing just what its camera is seeing. I realize the scene won't be as tall as the Oculus wants (high resolution) but the effect may still be good.

**Additional context**
Add any other context or screenshots about the feature request here.
I will implement this with head tracking that sends commands to the robot's pan and tilt camera mount.

peifeng-unity, 2021-11-08:
Thanks for reaching out. I have created an internal ticket (AIRO-1512) and we will follow up with you.

LaurieCheers-unity, 2021-11-08:
Hi, you can certainly do this in Unity. The simple approach would be to create two textured quads, one to display the left images and one for the right. Then make two separate camera objects, one for each eye. Set one quad to be visible only to the left camera, and one to the right camera.  https://answers.unity.com/questions/1534621/vr-how-to-render-gameobject-in-one-eye-only.html

If you parent each quad to the corresponding camera, you can scale and position them so that they always fill your view... but I wouldn't recommend it. This probably won't work very well - the ability to change your view by turning your head is fairly essential for a comfortable VR experience. You'd probably find more success leaving the textured quads in world space, so that it feels like you're watching the robot's view on a 3d TV screen.

For a fancier version you could make a Single Pass shader that checks STEREO_EYE_ID and displays one of two different textures accordingly. https://docs.unity3d.com/Manual/SinglePassStereoRendering.html

Russ76, 2021-11-09:
Wow, that sounds great, and complicated. I'll confess I'm only beginning to program in Unity! I found a repo that has the camera plugged directly into the Oculus, so that covers some of the difficulty. 
https://github.com/GeorgeAdamon/quest-realsense
I wondered if using ROS would help, or if that would slow down things too much, with images over the network.
I will work on it, thanks!

LaurieCheers-unity, 2021-11-09:
It looks like that repo is already taking care of the image streaming for you, so if it was up to me I would probably try to just use that - but yes, you could certainly use our ROS connection to receive and interpret textures or other data from your camera.
For example, this function can convert a ROS image message into a Unity texture for you: https://github.com/Unity-Technologies/ROS-TCP-Connector/blob/7f055108f43ab11ff297175a52af5d30fbac9d57/com.unity.robotics.ros-tcp-connector/Runtime/MessageGeneration/MessageExtensions.cs#L559

Russ76, 2021-11-10:
It handles image streaming if the camera is plugged into headset directly. That won't work with the robot, it will have to be over Wi-Fi. But thank you, with this help I think I can arrange it. The Quest 2 unit is impressive!

Russ76, 2021-11-29:
Project is not completed but we are looking at Zed camera now for streaming into Oculus headset instead of Realsense.

closed 2021-11-29 by Russ76