Unity-Technologies/ROS-TCP-Connector #234: Cant switch to ROS1 in the settings

timmermansjoy, 2021-11-09:
**Describe the bug**
When trying the [ROS SLAM example](https://github.com/Unity-Technologies/Robotics-Nav2-SLAM-Example) and switching form ROS2 to ROS1 in the settings it gets stuck at recompiling and it shows some errors as seen in the screenshot. This project also has to start up in safe mode afterwards

**To Reproduce**
Steps to reproduce the behavior:
1. Clone the project shows above with submodules
2. Open in unity
3. Switch ROS2 to ROS1
4. See error

**Console logs / stack traces**
see screenshot

**Expected behavior**
Changing from ROS2 to ROS1

**Screenshots**
[image]

**Environment (please complete the following information, where applicable):**
2 Computers:
- Unity Version: [Unity 2020.3.21f1, 2020.3.11f1 ]
- Unity machine OS + version: [MacOS Monterey, Ubuntu 20.04]
- ROS machine OS + version: [Docker]
- ROS–Unity communication: [Docker]
- Branch or version: [v0.6.0 (latest)]

**Additional context**
/

uninterrupted7, 2021-11-11:
ı think, it just only supported for ROS2

timmermansjoy, 2021-11-11:
@uninterrupted7 not true. you can know this simply by the fact that there is a button to switch from versions. and if you want more proof you can look at [Unity-Robotics-Hub](https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/ros_unity_integration/setup.md) documentation and see that they have it for ROS - 1 and 2

sdiao, 2021-11-15:
Hi @timmermansjoy , Thanks for reaching out. The [Robotics-Nav2-SLAM-Example](https://github.com/Unity-Technologies/Robotics-Nav2-SLAM-Example) is intended to be used with ROS2 only. Switching from ROS 2 to ROS won't work. 

However, you can switch between ROS and ROS2 in other tutorials such as [ROS Unity Integration](https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/ros_unity_integration/setup.md) or [Pick and Place](https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/pick_and_place/README.md) . 

Let us know if you have further questions. Thanks.

closed 2021-12-15 by github-actions[bot]