Unity-Technologies/ROS-TCP-Connector #246: Incorrect step size for Image message

Macbull, 2021-12-21:
**Issue and Steps to Reproduce** 
Adding a camera publisher to  unity project from https://github.com/Unity-Technologies/Robotics-Nav2-SLAM-Example using 
`ImageMsg rosImage = mainCameraTexture.ToImageMsg(header);`
causes error while using rqt or image_view node.
`ImageView.callback_image() while trying to convert image from 'rgba8' to 'rgb8' an exception was thrown (Image is wrongly formed: step < width * byte_depth * num_channels  or  4 != 691 * 1 * 4)
`

I think it is due to incorrect size here in this line of code, which just sends step=4 irrespective of width of image.
https://github.com/Unity-Technologies/ROS-TCP-Connector/blob/7f055108f43ab11ff297175a52af5d30fbac9d57/com.unity.robotics.ros-tcp-connector/Runtime/MessageGeneration/MessageExtensions.cs#L747

Problem can be solved by manually assigning step like 
`image_step = 4;`
`img_msg.step = image_step * (uint) frame_width;` 

I will try to reproduce and fix error for different texture types, And open a PR with correct code of `ToImageMsg()`. Although I am not sure if this is specific to ros2 or a general bug.

vidurvij-Unity, 2022-01-06:
Thank you for brining this to our attention and for your contribution. We will look into this pull request. [AIRO-1648]

mrpropellers, 2022-01-13:
Closing since fix is now merged into dev

PR #247 (2022-01-12): AIRO-1648 Fix step size in ToImageMsg() Function
## Proposed change(s)

Fixes incorrect step size for image message in ToImageMsg() function, which was causing error while trying to view Image topic in rqt, image_view and rviz.

### Useful links (GitHub issues, JIRA tickets, forum threads, etc.)

Closes #246 

### Types of change(s)

- [x] Bug fix
- [ ] New feature
- [ ] Code refactor
- [ ] Documentation update
- [ ] Other (please describe)

## Testing and Verification

Manually checked that image shows up in rqt for each format of texture. 

### Test Configuration:
- Unity Version: [Unity 202 [... cut]
closed 2022-01-13 by mrpropellers