Unity-Technologies/ROS-TCP-Connector #251: PointCloud2 Visibility in VR

jabbershort, 2022-01-26:
**Describe the bug**
I have suceeded in using the new Visualisations package to plot a pointcloud within my scene, however when viewing the scene in VR, the point cloud is invisible in both the 'Game' view and within the HMD. However, when looking at the 'Scene' view, I can see the point cloud.

**To Reproduce**
Docker container running pointcloud from .bag file and ROS-TCP-Connector. Unity 2020.3.26f1 with Visualisations package. When viewing in VR mode, the pointcloud topic can be selected in the topics list, and viewable in 'Scene' view but not in HMD or 'Game' view.

**Console logs / stack traces**
No error messages.

**Expected behavior**
Point cloud 'dots' visible in VR.

**Screenshots**
If applicable, add screenshots or videos to help explain your problem.

**Environment (please complete the following information, where applicable):**
- Unity Version: 2020.3.26f1
- Unity machine OS + version: Windows 10
- ROS machine OS + version: Dockerised ROS Noetic
- ROS–Unity communication: ROS-TCP-Endpoint & Connector

at669, 2022-02-11:
Hi, thanks for the post! I've created an internal ticket to investigate this, and the team will get back to you with updates. 

[Ticket#: AIRO-1711]

jabbershort, 2022-02-12:
Hi @at669,

Thanks for getting back to me, I ended up creating a custom message decoder and visualisation using particle systems, but would still be interested to compare to the visualisation package if it can work in VR.

maxilar20, 2023-06-30:
It's not a perfect fix but I changed the VR render mode to Multi-pass and it fixed it at the cost of performance