Unity-Technologies/ROS-TCP-Connector #261: Unable to install ROS-TCP-Connector

h8476532, 2022-02-21:
**Describe the bug**
I just follow the tutorial from：https://github.com/Unity-Technologies/ROS-TCP-Connector, want to install the ROS-TCP-Connector and use the Add package from git URL.... to install it, but it jumps out the error, and I also try to add the version like v0.4.0, v0.5.0, v0.6.0, v0.7.0, it still unable to install.
And I use the student version, so I don't know that is the student version can't download by URL or something.

**Screenshots**
[image]

**Environment (please complete the following information, where applicable):**
- Unity Version: Unity 2020.2.0f1
- Unity machine OS + version: Windows 10

h8476532, 2022-02-25:
Already know where is my fault.

closed 2022-02-25 by h8476532

XiaoLaoDi, 2022-06-01:
@h8476532 hi, same problem here, would you mind share how to solve the problem ?