Unity-Technologies/ROS-TCP-Connector #267: Namespace issues with using Unity.Robotics.Visualizations

bryceikeda, 2022-03-14:
**Describe the bug**
When I try to create my own custom visualizations, I am getting namespace errors. I'm not sure how to properly include the dll to reference the Robotics Visualization package. 

**To Reproduce**
1. Import packages URDF Importer, ROS-TCP-Connector and Robotics Visualizations
2. Create a new script using examples from [Visualizations.md](https://github.com/Unity-Technologies/ROS-TCP-Connector/blob/main/com.unity.robotics.visualizations/Visualizations.md). 
3. Change namespace from using Unity.Robotics.MessageVisualizers to using Unity.Robotics.Visualizations

**Console logs / stack traces**
```
The type or namespace name 'Visualizations' does not exist in the namespace 'Unity.Robotics' (are you missing an assembly reference?)
```

**Expected behavior**
I should be able to reference the functions within the namespace Unity.Robotics.Visualizations

**Environment (please complete the following information, where applicable):**
- Unity Version: [e.g. Unity 20203.28f1]
- Windows 11
- v0.7.0

at669, 2022-03-14:
Thanks for the post! You'll need to edit or create your [asmdef](https://docs.unity3d.com/Manual/ScriptCompilationAssemblyDefinitionFiles.html#create-asmdef) in the scripts folder you're working out of to include the `Unity.Robotics.Visualizations` reference. Hope this helps!

bryceikeda, 2022-03-14:
@at669 Of course... it worked, thanks for the help!

closed 2022-03-14 by bryceikeda