Unity-Technologies/ROS-TCP-Connector #280: ROS TCP Connector doesn't work on my Oculus Quest 2

jbg73, 2022-04-28:
Hi Everyone,

I'm trying to finish my end-of-degree project, which consists of doing some VR tasks with NAO Robot on Oculus Quest 2 and Unity. To do that, I have some scripts in ROS, so my aim is to connect Unity and ROS. So far I have managed to connect Unity and ROS successfully using ROS-TCP-Connector. However, now that I'm trying to run the app on the Oculus Quest 2, after I build and run it, nothing happens. I can see the scene with the robot and I get no compilation errors, however, it seems that the connection between Oculus and ROS never happens. 

I tried another thing in order to see if ROS was the problem. I created a script in python which opens a server and waits for Unity to act as client. With this same ROS-TCP-Connector, when I specify the server ip (the one I created in python) and I run the app in Unity, I can see in the terminal that something connected to the server. However, again, when I build it and run it from the Oculus, the server stays waiting because nothing connects to it.

I don't know if any of you can help or advice me on how to continue. I have searched on the internet and I have tried a lot of things but any of them worked at all.

Thank you so much in advance!

ndaley7, 2022-08-04:
Its  a somewhat basic question, but if it is possible, are you able to ping the ros server IP address from the quest?

Stan850, 2022-08-05:
Hi all,

I have also experienced the same connection issue with both the Quest 1 and Quest 2 when trying to connect my app (when built onto the device) to a ROS instance running on my computer. I am linking [https://github.com/Unity-Technologies/Unity-Robotics-Hub/issues/359](url) which contains an exhaustive list of things I have tried (not sure if @jbg73 had tried them as well).