Unity-Technologies/ROS-TCP-Connector #281: ActionServer and ActionClient ROS Support

ptiwari0664, 2022-05-10:
Feature Request: At present Action server and client modules are not supported in the ros unity connection. As a result of this, many manipulator services are not enabled.

sarah-gibson, 2022-05-11:
@ptiwari0664  Thank you for this request. It is on our radar and in our backlog.

scottwillmoore, 2022-05-13:
For a recent research project, we ended up developing a very simple implementation of a ROS action client. I doubt it is feature complete, or entirely correct but you can find it in our repository: https://github.com/scottwillmoore/fetch_vr. The implementation is in the `unity/Assets/Fetch VR/Scripts/Robotics` folder.

In case you try to use our implementation, the current messages generated by ROS TCP Connector for ROS actions were not ideal, as the they were not generic, which made it a little more difficult to develop a reusable action client. I ended up using reflection to implement my own feedback/goal/result messages to workaround this issue. It is important to note that this does require manual registration of your action messages as seen in `unity/Assets/Fetch VR/Scripts/Extensions/ROSExtensions.cs`.

Ushtarador, 2022-11-23:
Hello @sarah-gibson, I wanted to ask if there is any progress on the issue, and if you can give any timeline on when the feature would be available? I'm currently using a rather ugly workaround with what is available, and Action Servers would be a much more appropriate solution for us.