Unity-Technologies/ROS-TCP-Connector #301: Quality of service setting missing and laser scan (/scan) on ground level

Fabulani, 2022-12-04:
Our robot uses quality of service 'RELIABILITY=VOLATILE' for the laser scan, but it needs to be changed to 'RELIABLE' to work with Unity. I couldn't find anywhere to configure it in Unity. Then, when it works, the laser scan is on ground-level. It should be a bit above it. 

The global and local costmaps also don't align. This, plus the laser scan issue, makes me think it might be a problem with the /tf.

When I run the [Nav2-SLAM-Example](https://github.com/Unity-Technologies/Robotics-Nav2-SLAM-Example), everything is aligned and properly positioned (with the data from Unity's simulated robot).

**Environment:**
- Unity Version: 2021.3.14f1
- Unity machine OS + version: Ubuntu
- ROS machine OS + version: Docker image running ROS-galactic (osrf/ros:galactic-desktop)
- ROS–Unity communication: Docker container running ROS-TCP-Endpoint

In the following image, pink is the floor on Y=-1, laser scan are the colored dots, white dots are the point cloud (unrelated), and the costmap is the white+black-ink square (also looks weird).
[image]

Nav2-SLAM-Example has maps and laserscan properly aligned and positioned:
[image]

jankolkmeier, 2023-03-22:
FIY, I just ran into the same issue, tried to work around it by changing the QoS in `ros_tcp_endpoint/subscriber.py` to `reliability=ReliabilityPolicy.BEST_EFFORT`. While that fixed the issue, paradoxically, it also fixed it when using the default ros_tcp_endpoint, so I'm not sure if it is really related to any QoS settings.

chwu-rwth, 2023-08-23:
@jankolkmeier Hi, I am having the issue where there is incompatibility issue with Unity and ROS. The laser scanner driver that we are using uses BEST EFFORT, but unity apparently only supports RELIABLE. I tried looking into the subscriber.py to change the QoS, but i cant seem to find the line of code of: 
`reliability=ReliabilityPolicy`

Do you mind double checking if that is still a valid solution?

KingKiger, 2023-08-29:
Change the qos_profile in `ros_tcp_endpoint/subscriber.py` like this: 
`qos_profile = QoSProfile(depth=queue_size, reliability=QoSReliabilityPolicy.BEST_EFFORT)`

Hope this helps.