Unity-Technologies/ROS-TCP-Connector #319: Support common interface messages in humble

jcsantamaria, 2024-05-07:
**Describe the bug**
The ROS2 distro Humble (https://docs.ros.org/en/humble/index.html) has updated [common_interface](https://github.com/ros2/common_interfaces/tree/humble) messages. The TCP Connector throws an exception when attempting to deserialize, for example, MarkerMsg because the message in Humble has new fields.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a project that subscribe to a MarkerMsg topic (or MarkerArray).
2. Publish a MarkerMsg message
3. Observe an exception thrown in MessageDeserializer.cs

**Console logs / stack traces**
Please wrap in [triple backticks (```)](https://help.github.com/en/articles/creating-and-highlighting-code-blocks) to make it easier to read.

**Expected behavior**
MarkerMsg (and MarkerArray) messages should desterilize without exceptions.

**Screenshots**
If applicable, add screenshots or videos to help explain your problem.

**Environment (please complete the following information, where applicable):**
- Unity Version: [e.g. Unity 2021.3.2f1]
- Unity machine OS + version: [e.g. Windows 10]
- ROS machine OS + version: [e.g. Ubuntu 20.04, ROS Humble]
- ROS–Unity communication: [e.g. Docker]
- Branch or version: [e.g. v0.7.0]

**Additional context**
Running MessageGenerator tool on the Humble branch of common_interfaces repo fixes the problem.

robotics-qc, 2025-04-10:
I had some issue with the header message but I was able to solve that by building the humble branch.

But for marker images it does not seem to work
There is an error in deserialization - which I assume is the error you have mentioned here

```
ArgumentOutOfRangeException: Non-negative number required.
Parameter name: count
System.Text.UTF8Encoding.GetString (System.Byte[] bytes, System.Int32 index, System.Int32 count) (at <13c0c460649d4ce49f991e2c222fa635>:0)
Unity.Robotics.ROSTCPConnector.MessageGeneration.MessageDeserializer.Read (System.String& value) (at ./Library/PackageCache/com.unity.robotics.ros-tcp-connector@bdad6c87bd9e/Runtime/MessageGeneration/MessageDeserializer.cs:158)
RosMessageTypes.Visualization.MarkerMsg..ctor (Unity.Robotics.ROSTCPConnector.MessageGeneration.MessageDeserializer deserializer) (at Assets/RosMessages/Visualization/msg/MarkerMsg.cs:152)
RosMessageTypes.Visualization.MarkerMsg.Deserialize (Unity.Robotics.ROSTCPConnector.MessageGeneration.MessageDeserializer deserializer) (at Assets/RosMessages/Visualization/msg/MarkerMsg.cs:136)
Unity.Robotics.ROSTCPConnector.MessageGeneration.MessageDeserializer.Read[T] (T[]& values, System.Func`2[T,TResult] loader, System.Int32 length) (at ./Library/PackageCache/com.unity.robotics.ros-tcp-connector@bdad6c87bd9e/Runtime/MessageGeneration/MessageDeserializer.cs:183)
RosMessageTypes.MoveitTaskConstructor.SolutionInfoMsg..ctor (Unity.Robotics.ROSTCPConnector.MessageGeneration.MessageDeserializer deserializer) (at Assets/RosMessages/MoveitTaskConstructor/msg/SolutionInfoMsg.cs:53)
RosMessageTypes.MoveitTaskConstructor.SolutionInfoMsg.Deserialize (Unity.Robotics.ROSTCPConnector.MessageGeneration.MessageDeserializer deserializer) (at Assets/RosMessages/MoveitTaskConstructor/msg/SolutionInfoMsg.cs:45)
RosMessageTypes.MoveitTaskConstructor.SubSolutionMsg..ctor (Unity.Robotics.ROSTCPConnector.MessageGeneration.MessageDeserializer deserializer) (at Assets/RosMessages/MoveitTaskConstructor/msg/SubSolutionMsg.cs:37)
RosMessageTypes.MoveitTaskConstructor.SubSolutionMsg.Deserialize (Unity.Robotics.ROSTCPConnector.MessageGeneration.MessageDeserializer deserializer) (at Assets/RosMessages/MoveitTaskConstructor/msg/SubSolutionMsg.cs:33)
Unity.Robotics.ROSTCPConnector.MessageGeneration.MessageDeserializer.Read[T] (T[]& values, System.Func`2[T,TResult] loader, System.Int32 length) (at ./Library/PackageCache/com.unity.robotics.ros-tcp-connector@bdad6c87bd9e/Runtime/MessageGeneration/MessageDeserializer.cs:183)
RosMessageTypes.MoveitTaskConstructor.SolutionMsg..ctor (Unity.Robotics.ROSTCPConnector.MessageGeneration.MessageDeserializer deserializer) (at Assets/RosMessages/MoveitTaskConstructor/msg/SolutionMsg.cs:47)
RosMessageTypes.MoveitTaskConstructor.SolutionMsg.Deserialize (Unity.Robotics.ROSTCPConnector.MessageGeneration.MessageDeserializer deserializer) (at Assets/RosMessages/MoveitTaskConstructor/msg/SolutionMsg.cs:41)
Unity.Robotics.ROSTCPConnector.RosTopicState.Deserialize (System.Byte[] data) (at ./Library/PackageCache/com.unity.robotics.ros-tcp-connector@bdad6c87bd9e/Runtime/TcpConnector/RosTopicState.cs:147)
Unity.Robotics.ROSTCPConnector.RosTopicState.OnMessageReceived (System.Byte[] data) (at ./Library/PackageCache/com.unity.robotics.ros-tcp-connector@bdad6c87bd9e/Runtime/TcpConnector/RosTopicState.cs:92)
Unity.Robotics.ROSTCPConnector.ROSConnection.Update () (at ./Library/PackageCache/com.unity.robotics.ros-tcp-connector@bdad6c87bd9e/Runtime/TcpConnector/ROSConnection.cs:600)
UnityEngine.Debug:LogException(Exception)
Unity.Robotics.ROSTCPConnector.ROSConnection:Update() (at ./Library/PackageCache/com.unity.robotics.ros-tcp-connector@bdad6c87bd9e/Runtime/TcpConnector/ROSConnection.cs:604)
```

Is there anything else you did, aside from rebuilding the messages?

hijimasa, 2025-07-01:
I think this bug is fixed by below PR.
https://github.com/Unity-Technologies/ROS-TCP-Connector/pull/322
Since updates to the repository have stopped, I am merging it into the dev-hijimasa branch of below repository.
https://github.com/hijimasa/ROS-TCP-Connector